nicman23/dkms-hid-nintendo #1: Module doesn't work together with Joycond

joaorb64, 2020-03-23:
First of all, I want to thank you for this dkms module. I got here from an issue you posted in the original module's repo.

I installed the module from here and I'm using it on Ubuntu 19.10, then I installed joycond. Joycond service is up and running, but for some reason it won't take control over the driver. If I connect a pro controller, I'm not required to press R+L to assign it as a controller and if I connect Joycons they are all assigned directly to singular joycons, not giving me a chance to join them with R+L.

Am I doing something wrong, or is this expected?

nicman23, 2020-03-23:
i have not tested this with joycons (only the pro controller). i ll test it and report back

DCNick3, 2020-04-01:
Can reproduce. `joycond` works with custom-compiled kernel, but not with `hid-nintendo-dkms` package from AUR.

joaorb64, 2020-04-04:
The #define did the trick! It's fully functional now. Also now Dolphin detects rumble for the controllers.

nicman23, 2020-04-04:
oh nice, i ll commit it right now

nicman23, 2020-04-04:
hmm for some reason, dolphin does not detect the rumble in my pro controller - but i do not have the joycond daemon installed

DanielOgorchock, 2020-04-04:
joycond shouldn't impact pro controller rumble functionality. Does fftest work with the pro controller for you?

joaorb64, 2020-04-06:
I tested using rumble on Dolphin and the intensity is too strong... to the point that I prefered turning it off.
I was also interested in using the gyro capabilities of the joycons, but from what I understood most emulators (Dolphin, Citra, Cemu) work with DSU Server Applications, which I guess should be developed as a separate application? Or is there already a way to use gyro input in emulators? (maybe going a bit off-topic here)

DanielOgorchock, 2020-04-06:
You can lower the intensity of the rumble like shown below.
[image]

I just pushed a new patch to joycond that updates the udev rules to make the IMU accessible by non-root users. You can then configure dolphin to use the accel/gyro on its motion input tab as shown below. Note you have to right click on the inputs to select from the list. This was done with the pro controller, but it should be the same for the joy-cons (though I'm not sure if dolphin will associate both IMUs with the joycond virtual device).
[image]

nicman23, 2020-04-06:
@DanielOgorchock updated the module to not check for the ff flag. also i had to uninstall joycond due to the /dev/input of both the controller and the gyro having too restrictive permissions (and i am not using joycons anyways).

DanielOgorchock, 2020-04-06:
joycond restricts the main input device's permissions until you "pair" the controller by pressing both triggers together (it will stop flashing the LEDs and restore permissions). The patch I just submitted fixes permissions for the IMU evdev.

nicman23, 2020-04-07:
@DanielOgorchock ah i see. also with the force feedback enabled, i am getting a lot more disconnects with dmesg reporting:

```
[ 2005.066174] nintendo 0005:057E:2009.0017: controller MAC = 7C:BB:8A:F9:25:B1
[ 2005.141685] input: Nintendo Switch Pro Controller as /devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-4/1-4.3/1-4.3:1.0/bluetooth/hci0/hci0:11/0005:057E:2009.0017/input/input59
[ 2005.141870] input: Nintendo Switch Pro Controller IMU as /devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-4/1-4.3/1-4.3:1.0/bluetooth/hci0/hci0:11/0005:057E:2009.0017/input/input60
[ 2015.248731] nintendo 0005:057E:2009.0017: Failed to set rumble; e=-110
[ 2015.675395] Bluetooth: hci0: command 0x0804 tx timeout
[ 2015.788728] nintendo 0005:057E:2009.0017: Failed to set rumble; e=-110
[ 2016.328765] nintendo 0005:057E:2009.0017: Failed to set rumble; e=-110
[ 2016.868763] nintendo 0005:057E:2009.0017: Failed to set rumble; e=-110
[ 2017.408770] nintendo 0005:057E:2009.0017: Failed to set rumble; e=-110
[ 2017.948742] nintendo 0005:057E:2009.0017: Failed to set rumble; e=-110

```

joaorb64, 2020-04-08:
I was messing up with motion input on Dolphin and was sad that I couldn't use motion in other emulators like Cemu, Citra, etc, so I researched for a bit and put this together: https://github.com/joaorb64/joycond-cemuhook

With this you can setup dual joycon buttons on Dolphin and have the motion input be automatically assigned without any extra configs, and also use the input for other emulators.

I also created this guide https://joaorb64.github.io/2020/04/08/using-nintendo-switch-controllers-on-linux.html that I intend to improve on soon with individual installing steps.

closed 2020-04-09 by joaorb64