nicman23/dkms-hid-nintendo #2: Nintendo Switch controller with Steam

Peque, 2020-04-25:
I installed the kernel module and the controller seems to pair successfully (the LED indicating controller 1 is on). ^^

However, I wonder how can I use it with Steam. It seems it is not recognized by it.

Am I missing something? Is there something I can do to check if the controller is being correctly identified by the system and is sending actions when I click buttons for example?

nicman23, 2020-04-25:
see https://wiki.archlinux.org/index.php/Gamepad#Using_hid-nintendo_with_Steam_Games

Peque, 2020-04-25:
@nicman23 Yeah, that didn't work. Any idea about how could I debug/fix it?

nicman23, 2020-04-25:
have you run controllermap? (same wiki article mentions it)

also do not have any other controller to test if any works with steam?

Peque, 2020-04-25:
@nicman23 I have not run `controllermap`. Should I?

I do have another controller (an X-box clone controller), which works just fine with Steam.

nicman23, 2020-04-26:
yes, just run it with sudo. also are non steam games / applications that are sdl based working at all?

Peque, 2020-04-26:
@nicman23 Tried with SuperTux, since that one seems to be using SDL. The controller is not working.

I will try to find `controllermap`. I am not using Arch, but Fedora 31. :sweat_smile: 

Is there any way I could `cat` or monitor a file/device to see if at least the OS is receiving any bytes from the controller? (Before any bytes-to-action mapping takes place)

nicman23, 2020-04-26:
nah man do not worry, just run controllermap with sudo and reboot

Peque, 2020-04-26:
@nicman23 It seems that is not helping either. :shrug:

v1993, 2020-05-04:
I'm having a similar issue (first LED light up but controller is not working) and have some potentially useful information. This shows up in `dmesg` when I connect to Pro controller:

```
[  126.214410] hid-generic 0005:057E:2009.0005: unknown main item tag 0x0
[  126.214472] input: Pro Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/bluetooth/hci0/hci0:67/0005:057E:2009.0005/input/input21
[  126.214638] hid-generic 0005:057E:2009.0005: input,hidraw4: BLUETOOTH HID v0.01 Gamepad [Pro Controller] on 00:1a:7d:da:71:13
[  126.322731] nintendo 0005:057E:2009.0005: unknown main item tag 0x0
[  126.322967] nintendo 0005:057E:2009.0005: hidraw4: BLUETOOTH HID v80.01 Gamepad [Pro Controller] on 00:1a:7d:da:71:13
[  128.528602] nintendo 0005:057E:2009.0005: controller MAC = 64:B5:C6:44:E2:09
[  128.573868] input: Nintendo Switch Pro Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/bluetooth/hci0/hci0:67/0005:057E:2009.0005/input/input22
[  128.574190] input: Nintendo Switch Pro Controller IMU as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/bluetooth/hci0/hci0:67/0005:057E:2009.0005/input/input23
```
Also, joystick does not get registered in the system:

```
v@v-home:~$ evdev-joystick --l
v@v-home:~$
```

v1993, 2020-05-04:
Alright, it looks like I was wrong: while `evdev-joystick --l` shows no joysticks, but two new event devices appear in `/dev/input` that `evdev-joystick` can easily test. `jstest-sdl` also identifies controller.

I'm having problems with drift and calibration. Should I open separate issue for that?

joaorb64, 2020-05-04:
Possibly related:  https://github.com/ValveSoftware/steam-for-linux/issues/6651

v1993, 2020-05-05:
@joaorb64 That issue should be covered by firejail workaround from arch wiki. But here joystick is not working in non-steam games too...

Virusmater, 2020-05-31:
@v1993 regarding calibration you can try out https://github.com/Virusmater/evdev-joystick-calibration

closed 2020-06-30 by nicman23