nicman23/dkms-hid-nintendo #8: Connexion problems of the joycon

Yamakaky, 2020-06-22:
I have some problems connecting my joycon to my linux install (no problem on windows on another computer). When I pair it, everything seems fine, but the joycon disconnects a few seconds later. If I turn it back on by clicking a button, the leds blinks a few seconds then turn off, like when the switch is off. In the bluetooth logs, I see the Joycon connecting then disconnecting, no more info I can find. Do you know what could cause this problem? A friend has the same issue.

I'm on archlinux / linux 5.7.

nicman23, 2020-06-30:
can you please try a different bt adapter?

Yamakaky, 2020-07-01:
Unfortunnatly no, I'll try to find one.

ssbvr, 2020-07-16:
I have a similar problem on a Raspberry Pi 3 / Retropie 4.6. The JoyCon connect and loose connection after a few seconds, they reconnect, work for a few seconds and disconnect.

nicman23, 2020-07-16:
are the lights blinking?

ssbvr, 2020-07-17:
The light pattern behaves has expected, it might not be the default one because I am using the joycon driver of Daniel Ogorchock https://github.com/DanielOgorchock/joycond to have both controllers detected as one. First the lights are are blinking sequentially (pairing mode), once they are connected they start flashing all together and after pressing L+R only the first light stays on and I can use them as a single controller. The unexpected behaviour is that the single controllers disconnect randomly and I have to reconnect them each 15/30 seconds. Though, for the first time I just got them connected for longer than one minute.

nicman23, 2020-07-17:
if you have a switch pro try connecting it with usb and see if the issue is the bluetooth

ssbvr, 2020-07-20:
I forgot to mention that I can connect the joycon with the grip through the cable and the connection last without any problem. Indeed, at the moment I am using the controller with the cable connection.

nicman23, 2020-08-01:
this is probably hw issues with the bluetooth adapter. also upstream tracks issues with bluetooth at https://github.com/DanielOgorchock/linux/issues/7

closing for now until upstream changes something

closed 2020-08-01 by nicman23