nicman23/dkms-hid-nintendo #12: Units for IMU values?

v1993, 2020-08-06:
What are units used for IMU values? I was able to guess values for accelerometer, but it is less trivial for gyro.

nicman23, 2020-08-06:
sorry i do not really know, this is a downstream project. please ask at https://github.com/DanielOgorchock/linux

DanielOgorchock, 2020-08-16:
Please see this new revision of the IMU patch which has cleaned things up and made them more clear: https://github.com/DanielOgorchock/linux/commit/4a14fc82298a36e70c5e9dc809cdff8c4f68517e

Most relevant to this question:
```
/*
 * The controller's accelerometer has a sensor resolution of 16bits and is
 * configured with a range of +-8000 milliGs. Therefore, the resolution can be
 * calculated thus: (2^16-1)/(8000 * 2) = 4.096 digits per milliG
 * Resolution per G (rather than per millliG): 4.096 * 1000 = 4096 digits per G
 * Alternatively: 1/4096 = .0002441 Gs per digit
 */
static const s32 JC_IMU_MAX_ACCEL_MAG		= 32767;
static const u16 JC_IMU_ACCEL_RES_PER_G		= 4096;
static const u16 JC_IMU_ACCEL_FUZZ		= 10;
static const u16 JC_IMU_ACCEL_FLAT		/*= 0*/;

/*
 * The controller's gyroscope has a sensor resolution of 16bits and is
 * configured with a range of +-2000 degrees/second.
 * Digits per dps: (2^16 -1)/(2000*2) = 16.38375
 * dps per digit: 16.38375E-1 = .0610
 *
 * STMicro recommends in the datasheet to add 15% to the dps/digit. This allows
 * the full sensitivity range to be saturated without clipping. This yields more
 * accurate results, so it's the technique this driver uses.
 * dps per digit (corrected): .0610 * 1.15 = .0702
 * digits per dps (corrected): .0702E-1 = 14.247
 *
 * Now, 14.247 truncating to 14 loses a lot of precision, so we rescale the
 * min/max range by 1000.
 */
static const s32 JC_IMU_PREC_RANGE_SCALE	= 1000;
/* Note: change mag and res_per_dps if prec_range_scale is ever altered */
static const s32 JC_IMU_MAX_GYRO_MAG		= 32767000; /* (2^16-1)*1000 */
static const u16 JC_IMU_GYRO_RES_PER_DPS	= 14247; /* (14.247*1000) */
static const u16 JC_IMU_GYRO_FUZZ		= 10;
static const u16 JC_IMU_GYRO_FLAT		/*= 0*/;
```

With the latest patch revision (not yet in this dkms repo), dividing by the resolution will give you degrees/second (I think that should be true for the prior version as well).

Another important thing to note about this patch is that it negates two of the right joy-con's axes to be consistent with the axes' orientations on the pro controller and left joy-con. The axes in question are Y and Z.

closed 2021-02-01 by v1993