nicman23/dkms-hid-nintendo #14: Error on Build

Mumrik93, 2020-08-27:
Been trying for about 40 min to install this but i keep getting the same error-message;

"Error! Could not find module source directory.
Directory: /usr/src/nintendo-1.0 does not exist."

I tried removing all the files and clone the git again, doing it all over, but i kept getting the same error again :( Being a noob I have to admit defeat and hope someone else knows what the issue might be caused by.

joaorb64, 2020-08-27:
If you open /usr/src/, is there a `nintendo-x` directory there?
My guess is that the current version creates nintendo-2.0 instead of 1.0, so try changing the version when running the commands.

Mumrik93, 2020-08-27:
Gave that a try and it did indeed solve the problem, thank you!

closed 2020-08-27 by Mumrik93