nicman23/dkms-hid-nintendo #24: Linux recognizes the controller, but no games do

j-james, 2020-11-13:
After installing this [AUR package](https://aur.archlinux.org/packages/hid-nintendo-dkms/), I'm unable to get my controller to work.

Although `lsusb` recognizes a `Nintendo Co., Ltd Switch Pro Controller`, the player-1 indicator light pops on, and `wine control joy.cpl` works well, no games I've tested this with or [the HTML5 GamePad API](https://gamepad-tester.com/) recognize it. For comparison, an Xbox 360 controller shows up and is usable in all of the above.

Is there something I'm missing, like a post-install command or similar? What seems strange to me is that `/dev/input` and elsewhere seem to contain all the right files...

nicman23, 2020-11-13:
all sdl games should support it. 

https://wiki.archlinux.org/index.php/Gamepad#Using_hid-nintendo_with_SDL2_Games

j-james, 2020-11-13:
Ah, thank you, I had that in the wrong spot.

closed 2020-11-13 by j-james