nicman23/dkms-hid-nintendo #26: Can't get my controller to pair

vjau, 2021-01-02:
I have installed the dkms driver 
```
dkms status
-> nintendo, 3.0, 5.4.0-58-generic, x86_64: installed
```
I have installed the joycond daemon
```
sudo systemctl status joycond
-> Loaded: loaded (/etc/systemd/system/joycond.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2021-01-02 20:38:24 CET; 14min ago
   Main PID: 5053 (joycond)
      Tasks: 1 (limit: 18826)
     Memory: 696.0K
     CGroup: /system.slice/joycond.service
             └─5053 /usr/bin/joycond
```

```
sudo journalctl -u joycond
->janv. 02 20:38:24 vincent-ThinkPad-T580 systemd[1]: Started joycond.
janv. 02 20:38:24 vincent-ThinkPad-T580 joycond[5053]: adding epoll_subscriber: fd=4

```

But when i press on both upper switches, nothing nintendo related shows up in the bluetooth devices list.
What to check next ?

Thank you.

nicman23, 2021-01-03:
to pair you need to long press the small dot button (which is normally covered by the sides of the switch) till the leds start to blink

closed 2021-01-03 by nicman23

vjau, 2021-01-03:
Than you, that works.