nicman23/dkms-hid-nintendo #28: Pro controller random disconnects with bluetooth on Streets of Rage 4

sirdeiu, 2021-01-30:
Hi,

Using the following:

openSUSE Tumbleweed with kernel: 5.10.9-1-default
KDE Plasma 5
hid-nintendo 3
joycond

While trying to play Streets of Rage 4, I keep having random disconnects from the Pro Controller, while it is connected via Bluetooth. I have tried it with two Bluetooth adapters (Intel Wifi AX200 and ASUS BT-400 (with firmware).

The controller seems fine when connected to PC with USB cable. Also, seeing this happen more random and often when trying the game via Steam and raised an issue there also.

While playing the game on Windows 10 with Steam, on the same PC and hardware, the issue is not present.

Any clues how to debug/test this further please?

nicman23, 2021-01-30:
does streets of rage use ramble a lot?

sirdeiu, 2021-01-30:
Game has rumble, yes. I would say rumble is quite active while playing. Game has Vibration Intensity option (0-100%). I've now tried setting that to 0% and playing and it seems to work, the controller was active while finishing a game level without disconnecting.

Then I also disabled rumble on Steam controller settings for the Pro, but there it still seems to case the issue, though not as often as before.

nicman23, 2021-01-31:
please report here https://github.com/DanielOgorchock/linux/issues/7

closed 2021-01-31 by nicman23