nicman23/dkms-hid-nintendo #31: Cant install the kernel module

Penty420, 2021-03-02:
Hello guys
I need some help with installing the kernel module.
When I run the given commands, I get an error saying that no dkms.conf file was provided. How can I fix this? Can I fix this?
Thats whats happening:
`[root@daniel-aspirea71541g dkms-hid-nintendo]# git clone https://github.com/nicman23/dkms-hid-nintendo
cd dkms-hid-nintendo

sudo dkms add .
sudo dkms build nintendo -v 3.0
sudo dkms install nintendo -v 3.0
Cloning into 'dkms-hid-nintendo'...
remote: Enumerating objects: 374, done.
remote: Counting objects: 100% (374/374), done.
remote: Compressing objects: 100% (244/244), done.
remote: Total 374 (delta 54), reused 340 (delta 31), pack-reused 0
Receiving objects: 100% (374/374), 677.51 KiB | 1.68 MiB/s, done.
Resolving deltas: 100% (54/54), done.
Error! DKMS tree already contains: nintendo-3.1
You cannot add the same module/version combo more than once.
Error! Could not locate dkms.conf file.
File: /usr/src/nintendo-3.0/dkms.conf does not exist.
Error! Could not locate dkms.conf file.
File: /usr/src/nintendo-3.0/dkms.conf does not exist.
[root@daniel-aspirea71541g dkms-hid-nintendo]# 
`

Sanaki, 2021-03-02:
It's 3.1 now. It should work if you use `-v 3.1`.

z0rti, 2023-01-10:
dkms remove nintendo -v 3.1
And it's 3.2 now