nicman23/dkms-hid-nintendo #33: 3.1 keep dropping connection

jcsalt, 2021-03-02:
3.0 connects stable, 3.1 only goog for a minnute then drop.

ubuntu 20.10

durteetr, 2021-03-03:
Having similar issues, connects for less than a minute than drop.

Linux Mint 18.3 Sylvia

Linux version 4.15.0-132-generic (buildd@lgw01-amd64-034) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.12)) #136~16.04.1-Ubuntu SMP Tue Jan 12 18:22:20 UTC 2021

nicman23, 2021-03-03:
if any commit is to blame it is probably https://github.com/nicman23/dkms-hid-nintendo/commit/4693d14c98151821dc067cd85f4109b70fe2a816. can you test without it (but care to have 3.1 in dkms.conf)

SGiuseppe, 2021-03-05:
same problem also can the previous version 

My system openSUSE Tumbleweed last update 05/03/2021 kernel 5.10.16-1-default #1 SMP Sat Feb 13 16:20:19 UTC 2021 (11381f3) x86_64 x86_64 x86_64 GNU/Linux

jcsalt, 2021-03-14:
seems my issue is for the joycons only (dual joycon as a pro controller), just use the pro controller works great.

durteetr, 2021-03-16:
yeah, I am using the pro controller, that's where I'm having these issues

nicman23, 2021-03-16:
@jcsalt that probably means a hardware issue

SGiuseppe, 2021-03-16:
In my possession I have only two joycons, and the problem seems to not be resolved. The Bluetooth connector is "plungable" for protocol 4 supported by the Linux kernel without problems. I hope these data help.

iMonZ, 2021-03-21:
I have the same issue while im using cemuhook

v1993, 2021-04-03:
I'd like to ask to make sure that your problem is indeed specific to 3.1 before posting here.

I can confirm that starting with 3.1 disconnects became way more common for me. They happened before, but now they are just ridiculous.

DanielOgorchock, 2021-07-25:
https://github.com/nicman23/dkms-hid-nintendo/pull/48 should help with reducing disconnects

mutlucan96, 2021-12-10:
Random joycons randomly dropping the connection on 3.2. Ubuntu 21.10