nicman23/dkms-hid-nintendo #41: [Question] How can I disable the home led of the nintendo switch pro controller?

Toadfield, 2021-06-18:
It annoys me so much,I want to permanently disable it.

nicman23, 2021-06-18:
the led is not setup from us. are you running steam?

Toadfield, 2021-06-18:
But on reddit someone said that it is somehow possible to force the led to be off with dkms-hid-nintendo.
I use steam,but the led is also on when I open non steam games.

nicman23, 2021-06-19:
steam takes over and turns on the led. default for this module is off

if you are using arch linux i have made a joycond fork for keeping steam off the controller joycond-nicman23-git

closed 2021-06-19 by nicman23

nfp0, 2022-02-22:
@Toadfield And if you want to use it with Steam, there's a setting on Steam Big Picture that allows you to control the brightness of the home LED. I believe you can totally disable it too.