nicman23/dkms-hid-nintendo #42: analog sticks don't work

Toadfield, 2021-06-19:
I tested it with dolphin,when I try to calibrate the analog stick,then it doesn't work.

nicman23, 2021-06-19:
can you please show a screenshot of the dolphin's controller setup

Toadfield, 2021-06-22:
I installed a new system and tested it,now the problem is that my nintendo switch pro controller doesn't get registered at all,when I look up the controller devices with an emulator or whatever.
This is my system information:
[image]

Toadfield, 2021-06-22:
I use it via usb btw

Toadfield, 2021-06-22:
I tried it with other applications,the controller doesn't get listed in them.

Toadfield, 2021-06-23:
When I do `lsusb -v` in the terminal and search for "switch",then I get:
```
Bus 003 Device 005: ID 057e:2009 Nintendo Co., Ltd Switch Pro Controller
```
And also when steam isn't running and I connect my switch pro controller via usb and then press a button,then the leds blinking and then they are off after some seconds and nothing works.

Toadfield, 2021-06-23:
I tested it,it only works with steam and cemu,and the other problem is that with those apps the blue home led is turned on,even though I use joycond-nicman23-git.

Toadfield, 2021-06-23:
When I test my switch pro controller after opening cemu with a program called sc-controller,then it is completely buggy,I don't press anything,but the program thinks that I press those buttons:
[image]

Toadfield, 2021-06-23:
I hope that you can help me fast,cause I just want to play secret of mana with my controller now.

Toadfield, 2021-06-24:
I installed the linux-headers and now the controller itself works,but not for sc-controller.

Toadfield, 2021-06-24:
And the home led still glows blue when I start some programs like cemu or ppsspp.

Toadfield, 2021-06-24:
I will close this issue and make two seperate ones for each of the issues.

closed 2021-06-25 by Toadfield