nicman23/dkms-hid-nintendo #43: [Question] Is there a way to turn the green led buttons on the buttom of or select that it only uses player 1 led?

Toadfield, 2021-06-20:
The green led buttons annoy me,I want that they are turned off.
And if that doesn't work,I want that only the player 1 led gets used.

nicman23, 2021-06-20:
install joycond

closed 2021-06-20 by nicman23

Toadfield, 2021-06-20:
I already installed [this package](https://aur.archlinux.org/packages/joycond-nicman23-git/),do how can i select that the green led's aren't on?

nicman23, 2021-06-20:
oh sorry that is just for making it to be player 1 2 3 4 (when you connect multiple controllers)

if you really are that bothered with it i can see if i can get you a patch when i got some time.

Toadfield, 2021-06-20:
But sometimes the player 2 led is blinking even though I only connected one controller,is that also intended?

nicman23, 2021-06-20:
nothing should be blinking after initial connect

can you send a video with that?

Toadfield, 2021-06-20:
I don't have a phone or cam,I can't make a video of that.
I don't mean that it is blinking after connecting,I mean that it is just green,but not only the player 1 led,but sometimes the player 2 led.

Toadfield, 2021-06-24:
sometimes the 2 player leds or 3 player leds or 4 player leds are showing up instead of just the 1 player led even though I only have one controller connected.