nicman23/dkms-hid-nintendo #45: My switch pro controller doesn't work with sc-controller

Toadfield, 2021-06-24:
[sc-controller](https://github.com/kozec/sc-controller) is a program that emulates xbox360 input for controllers that aren't xbox360.
I want to use that for my switch pro controller cause many games only work with the xbox360 controller.
The problem is that I just can't add my controller,it doesn't show up:
[image]
It is not a sc-controller problem,cause that supports hid devices.
I also checked,the switch controller is connected via usb and works with things like emulators.

Toadfield, 2021-06-26:
I use it via usb by the way.

nicman23, 2021-06-29:
https://github.com/DanielOgorchock/linux

closed 2021-06-29 by nicman23