nicman23/dkms-hid-nintendo #50: Pro Controller y-axes and drift problem

slobodan-rs, 2021-11-12:
Hello,

I am having an odd problem with hid driver. The controller is working fine wired but on bluetooth I have a "calibration" problem. Immediately upon connecting the calibration with sticks is fine but if you press any other button on controller, other then sticks it messes up axes so "UP" doesn't go all the way and center drifts (the video explains it better). The numbers in calibration tool goes all over the place

Arch Linux with kernel 5.14.16 Try it also on different distros and linux kernels if that helps, the problem persists..

[https://reddit.com/link/qrwdke/video/9chgk9mui1z71/player]

Thank you!

nicman23, 2021-11-13:
it know this probably sucks but could you please try a different bt adapter?

slobodan-rs, 2021-11-15:
Ok. I will try in couple of days and I will let you know, just to get my hands on one.

nicman23, 2021-11-15:
sure, i only ask because these issues are not happening on usb.

slobodan-rs, 2021-11-15:
Yeah, very odd..

TheZipBon, 2022-02-11:
Any updates on this?

nicman23, 2022-02-12:
sorry but only solution i can give is a deadzone :/

TheZipBon, 2022-02-13:
Oh well. I would be glad if a proper fix was created, though, as wired has the limitation of making motion controls unwieldy.

nfp0, 2022-02-14:
It's weird that you're getting this behavior only on Bluetooth.
My Pro Controller does not have that problem on Bluetooth. Is your Pro Controller updated to the latest firmware version?

slobodan-rs, 2022-02-22:
Sorry for late reply. It acted same on different bluetooth, after kernel upgrade 5.15 everything is fine. Cannot recall if i updated firmware on controller in meantime but I doubt.