nicman23/dkms-hid-nintendo #51: System freeze after connecting 2 joy cons (Nintendo Switch)

jagiella, 2021-11-19:
My system is an ubuntu 22.04 jammy jellyfish (developement release) with gnome desktop. 

I tried the hid-nintendo on:
1. mainline kernel 5.15 with dkms-hid-nintendo as kernel extension (from your repository) 
2. mainline kernel 5.16-rc1 with hid-nintendo built into the kernel (to my knowledge still based on your code)

I can pair and connect the joy cons via bluetooth. Unfortunately after having 2 joy cons (left and right) connected the system will eventually freeze the desktop. The mouse cursor is still moving, but keyboard is unresponsive (even changing tty with ctrl+alt+F1...F8) and everything is frozen. I am forced to unplug power to shutdown/reboot the system.

By the way, I tried that before with 8bitdo sf30 pro controller in switch mode and its successfully contecting as switch pro controller and not hanging up. But with original joy cons from Nintendo it persistently freezes.

Also installed the joycond (and joycond-cemuhook).  

As the issue even persists in the new mainline kernel 5.16-rc1 I felt the need to warn you about this issue as it might cause problems for others as well in the future using that kernel.

nicman23, 2021-11-20:
Hey thanks for the report, this however is a downstream project and this is mostly not my code. Can you please report it to https://github.com/DanielOgorchock/linux/ ?

Also can you please attach dmesg ? Thanks

jagiella, 2021-11-20:
Okay I tried to reproduce the issue and this time it did not freeze, but the bluetooth ended up not being usable anymore.

What did I do?
- The 2 joy cons were already paired from a previous session.
- I switched both joy cons on.
- Both connected: LEDs showed joycon ids (left=1 LED, right=2 LEDs) and the bluetooth settings of ubuntu showed both as connected.

Whats the issue?
- None of the controllers were usable / visible as input devices.
- Deleting the joycons (for the purpose of repairing) didn't work anymore.
- The whole bluetooth setting menu is not working anymore:
  - impossible to disable or enable bluetooth 
  - impossible to remove the joy cons
- rebooting from commandline (sudo reboot) leads to a frozen system as initially discribed in the ticket. Needs a hardware reset.  

`
-- Journal begins at Sat 2021-11-06 20:42:51 CET, ends at Sat 2021-11-20 15:08:44 CET. --
Nov 20 14:52:05 nick-G501JW kernel: microcode: microcode updated early to revision 0x28, date = 2019-11-12
Nov 20 14:52:05 nick-G501JW kernel: Linux version 5.16.0-051600rc1-generic (kernel@kathleen) (gcc (Ubuntu 11.2.0-10ubuntu1) 11.2.0, GNU ld (GNU Binutils for Ubuntu) 2.37) #202111142330 SMP PREEMPT Sun Nov 14 23:35:10 UTC 2021
Nov 20 14:52:05 nick-G501JW kernel: Command line: BOOT_IMAGE=/boot/vmlinuz-5.16.0-051600rc1-generic root=UUID=89c7edd7-4bc5-4664-97d5-947d91d5a4cf ro quiet splash vt.handoff=7
Nov 20 14:52:05 nick-G501JW kernel: KERNEL supported cpus:
Nov 20 14:52:05 nick-G501JW kernel:   Intel GenuineIntel
Nov 20 14:52:05 nick-G501JW kernel:   AMD AuthenticAMD
Nov 20 14:52:05 nick-G501JW kernel:   Hygon HygonGenuine
Nov 20 14:52:05 nick-G501JW kernel:   Centaur CentaurHauls
Nov 20 14:52:05 nick-G501JW kernel:   zhaoxin   Shanghai  
Nov 20 14:52:05 nick-G501JW kernel: x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
Nov 20 14:52:05 nick-G501JW kernel: x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
Nov 20 14:52:05 nick-G501JW kernel: x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
Nov 20 14:52:05 nick-G501JW kernel: x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
Nov 20 14:52:05 nick-G501JW kernel: x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
Nov 20 14:52:05 nick-G501JW kernel: signal: max sigframe size: 1776
Nov 20 14:52:05 nick-G501JW kernel: BIOS-provided physical RAM map:
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x0000000000059000-0x000000000009dfff] usable
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x000000000009e000-0x000000000009ffff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x0000000000100000-0x00000000728e9fff] usable
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x00000000728ea000-0x00000000728f0fff] ACPI NVS
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x00000000728f1000-0x000000007321afff] usable
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x000000007321b000-0x00000000734b8fff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x00000000734b9000-0x000000007a012fff] usable
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x000000007a013000-0x000000007a209fff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x000000007a20a000-0x000000007a3f7fff] usable
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x000000007a3f8000-0x000000007ab00fff] ACPI NVS
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x000000007ab01000-0x000000007affefff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x000000007afff000-0x000000007affffff] usable
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x000000007bc00000-0x000000007fdfffff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x00000000f0000000-0x00000000f7ffffff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x00000000fed00000-0x00000000fed03fff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x0000000100000000-0x000000047f1fffff] usable
Nov 20 14:52:05 nick-G501JW kernel: NX (Execute Disable) protection: active
Nov 20 14:52:05 nick-G501JW kernel: e820: update [mem 0x724a1018-0x724cb857] usable ==> usable
Nov 20 14:52:05 nick-G501JW kernel: e820: update [mem 0x724a1018-0x724cb857] usable ==> usable
Nov 20 14:52:05 nick-G501JW kernel: extended physical RAM map:
Nov 20 14:52:05 nick-G501JW kernel: reserve setup_data: [mem 0x0000000000000000-0x0000000000057fff] usable
Nov 20 14:52:05 nick-G501JW kernel: reserve setup_data: [mem 0x0000000000058000-0x0000000000058fff] reserved
Nov 20 14:52:05 nick-G501JW kernel: reserve setup_data: [mem 0x0000000000059000-0x000000000009dfff] usable
Nov 20 14:52:05 nick-G501JW kernel: reserve setup_data: [mem 0x000000000009e000-0x000000000009ffff] reserved
Nov 20 14:52:05 nick-G501JW kernel: reserve setup_data: [mem 0x0000000000100000-0x00000000724a1017] usable
[... cut ...]
Nov 20 14:58:29 nick-G501JW kernel: ntfs3: Read-only LZX/Xpress compression included
Nov 20 14:58:29 nick-G501JW kernel: ntfs3: Unknown parameter 'windows_names'
Nov 20 15:01:01 nick-G501JW kernel: Bluetooth: hci0: command 0x0804 tx timeout
Nov 20 15:01:01 nick-G501JW kernel: Bluetooth: hci0: failed to restart LE scan: status 31
Nov 20 15:01:03 nick-G501JW kernel: Bluetooth: hci0: failed to disable LE scan: status 0x1f
Nov 20 15:02:07 nick-G501JW kernel: Bluetooth: hci0: HCI reset during shutdown failed
`

CactiChameleon9, 2022-04-11:
I have this issue too! However, for me a single joycon is the problem (my left one) and the right one connects fine as normal (if connected first). Do you always connect them in the same order?

My system sometimes crashes immediately or sometimes has a bit of a wait until things break (enough time to connect the good controller)

I will attach my dmesg in a bit