nicman23/dkms-hid-nintendo #53: can't connect, permission denied

z0rti, 2021-12-28:
i'm using debian 11 with kernel 5.10
when i try to connect the switch pro controller with blueman or 'bt-device -c  70:F0:xx:xx:xx:9D' i get this in the syslog:

```
Dec 28 18:44:11 zort bluetoothd[9953]: profiles/input/device.c:control_connect_cb() connect to 70:F0:xx:xx:xx:9D: Host is down (112)
Dec 28 18:44:41 zort bluetoothd[9953]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
Dec 28 18:44:42 zort bluetoothd[9953]: profiles/input/device.c:control_connect_cb() connect to 70:F0:xx:xx:xx:9D: Permission denied (13)
Dec 28 18:44:56 zort bluetoothd[9953]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
Dec 28 18:44:57 zort bluetoothd[9953]: profiles/input/device.c:control_connect_cb() connect to 70:F0:xx:xx:xx:9D: Permission denied (13)
Dec 28 18:47:37 zort bluetoothd[9953]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info

```

(when i press the bt button on the back)

does the controller reply with 'denied' so that probably means it doesn't want to be used with linux?

Sanaki, 2021-12-29:
Your user probably isn't a member of the plugdev group. Add the group and restart.

z0rti, 2021-12-29:
ok now i added the user to plugdev, restarted, but i still get permission denied.
a bt mouse i also got does work btw.

nicman23, 2021-12-31:
what distro?

z0rti, 2021-12-31:
debian 11 with kernel 5.10

nicman23, 2022-01-01:
debian might have a different equivalent user. have you installed joycond?

z0rti, 2022-01-02:
which /dev does it need to access, then i could look up what group i need?
yes joycond is installed (git), sorry i forgot to mention that.
is it normal that without (a working, running) joycond one gets 'permission denied'?
then it's probably an issue of joycond not hid-nintendo

nicman23, 2022-01-02:
it is. this driver does not handle permissions at all. look at the joycond udev rules.

z0rti, 2022-02-17:
Yes that was the problem, the install did not cp the udev rules over, now it works.

closed 2022-02-17 by z0rti