nicman23/dkms-hid-nintendo #57: 8BitDo controller is only recognized upon first sync

jonathansharpe, 2022-01-29:
I've been using my 8BitDo SN30Pro with Dolphin, using joycond-cemuhook-git, and dkms-hid-nintendo. When I first sync it to my computer as a Bluetooth device, it's recognized fine by both software. However, when I power off the controller and power it back on to reconnect, it's no longer recognized. joycond-cemuhook-git just says "Looking for Nintendo Switch controllers...", and it's also not recognized as a device in Dolphin. I'm assuming this is a driver issue, so I decided to post it here.

OS: Manjaro KDE 5.23

floydfire, 2022-02-14:
I think it's how the controller works, i have the same issue with the pro 2 and this is on both windows and linux, you have to reconnect everytime  (for me i just use the reset button) sadly i haven't found a fix. This is if i have it in switch mode, haven't tried on other modes, but i know for certain that in switch mode does that even on windows...

KastenKlicker, 2022-04-23:
I do have the same problem with my Joy Cons on Kubuntu 21.04 and Ubuntu 22.04. Didn't used them for something to this moment, I just connected my right one, powered my laptop off and on. Removing the Joy Con from the Known Devices list and reconnect it does work for me.

floydfire, 2022-04-24:
yea it may be something related to how the nintendo sync? cause i don't think the 8bitdo does the same with xinput or dinput

nicman23, 2022-05-12:
do you have joycond installed?

KastenKlicker, 2022-05-30:
It doesn't work correctly with or without joycond.

ali1234, 2022-12-30:
Steps to reproduce:

1. Turn off the controller
2. Hold Y and press start
3. Press the pairing button on the top of the controller
4. Start a bluetooth scan on the computer and connect the controller

Result: the controller is detected and the evdevs are created and work:
```
[1357797.440432] nintendo 0005:057E:2009.006B: unknown main item tag 0x0
[1357797.440686] nintendo 0005:057E:2009.006B: hidraw6: BLUETOOTH HID v80.01 Gamepad [Pro Controller] on 5c:f3:70:6b:90:f0
[1357799.521029] nintendo 0005:057E:2009.006B: using factory cal for left stick
[1357799.554824] nintendo 0005:057E:2009.006B: using factory cal for right stick
[1357799.657311] nintendo 0005:057E:2009.006B: using factory cal for IMU
[1357799.829867] nintendo 0005:057E:2009.006B: controller MAC = E4:17:D8:50:2A:0A
[1357800.373981] nintendo 0005:057E:2009.006B: Failed to set home LED dflt; ret=-110
[1357800.373997] nintendo 0005:057E:2009.006B: Failed to create leds; ret=-110
[1357800.374539] input: Nintendo Switch Pro Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/bluetooth/hci2/hci2:12/0005:057E:2009.006B/input/input185
[1357800.374998] input: Nintendo Switch Pro Controller IMU as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/bluetooth/hci2/hci2:12/0005:057E:2009.006B/input/input186
```

5. Turn off the controller
6. Turn on the controller
7. It will automatically reconnect, because it is paired

Result: controller is not initialized properly:

```
[1357847.938690] nintendo 0005:057E:2009.006C: unknown main item tag 0x0
[1357847.939016] nintendo 0005:057E:2009.006C: hidraw6: BLUETOOTH HID v80.01 Gamepad [Pro Controller] on 5c:f3:70:6b:90:f0
[1357852.031882] nintendo 0005:057E:2009.006C: failed reading SPI flash; ret=-110
[1357852.031903] nintendo 0005:057E:2009.006C: using factory cal for left stick
[1357854.079953] nintendo 0005:057E:2009.006C: failed reading SPI flash; ret=-110
[1357854.079968] nintendo 0005:057E:2009.006C: using factory cal for right stick
[1357856.128016] nintendo 0005:057E:2009.006C: failed reading SPI flash; ret=-110
[1357856.128032] nintendo 0005:057E:2009.006C: Failed to read left stick cal, using dflts; e=-110
[1357858.176097] nintendo 0005:057E:2009.006C: failed reading SPI flash; ret=-110
[1357858.176107] nintendo 0005:057E:2009.006C: Failed to read right stick cal, using dflts; e=-110
[1357860.228178] nintendo 0005:057E:2009.006C: failed reading SPI flash; ret=-110
[1357860.228194] nintendo 0005:057E:2009.006C: using factory cal for IMU
[1357862.272248] nintendo 0005:057E:2009.006C: failed reading SPI flash; ret=-110
[1357862.272263] nintendo 0005:057E:2009.006C: Failed to read IMU cal, using defaults; ret=-110
[1357862.272270] nintendo 0005:057E:2009.006C: Unable to read IMU calibration data
[1357864.320324] nintendo 0005:057E:2009.006C: Failed to set report mode; ret=-110
[1357864.320693] nintendo 0005:057E:2009.006C: probe - fail = -110
[1357864.320722] nintendo: probe of 0005:057E:2009.006C failed with error -110
```

In order to make the controller work again, you have to press the pairing button and re-pair it in the bluetooth manager.

ali1234, 2022-12-30:
With the `ignore_init_failure` option, the evdevs are created but never produce any events.

ali1234, 2022-12-30:
I have now tested firmware 1.38 and it behaves identically. I also tested 1.36 and older firmwares, but with these firmwares the controller does not function at all. It won't even turn on. Controller was shipped with 1.38.

ali1234, 2022-12-30:
Also note that this problem doesn't happen with hid-generic. The controller works fine after turning it off and on again. All buttons and axes work, but motion controls are not supported.

Adesin-fr, 2023-01-04:
I second this.
I've experienced the same issue !
No fixes found...
I've noticed also that when plugged via USB, it works (reports events)