nicman23/dkms-hid-nintendo #58: Pro controller weird issue

jnk19, 2022-02-20:
I am using pro controller with retroarch and when I have rumble enabled after sometime it disconnects. Connect again and after 2 minutes do the same.

It does not do that when the rumble is turned off. I have tried everything but still the same.

Any suggestion would be helpful

Regards..

nicman23, 2022-02-20:
unfortunately that happens with bluetooth. only thing that has worked for me was to get a better usb bluetooth. :C

closed 2022-02-20 by nicman23

nfp0, 2022-04-23:
@nicman23 Could you please specify what Bluetooth dongle you got that doesn't get affected by this issue?
As I described on this [this post](https://github.com/DanielOgorchock/linux/issues/33#issuecomment-1098360786), all my 5 Bluetooth dongles (4 of them officially certified) have this issue. All exactly with the same behavior. And also tested on different computers.
So it seems your working Bluetooth seems to be a very isolated case. Or maybe some other thing on your system is configured differently.

nicman23, 2022-05-12:
i just bought a cheap 5.0 tp link one

nfp0, 2022-05-12:
Is it the same `ID 2357:0604 TP-Link TP-Link UB500 Adapter` I have?

nicman23, 2022-05-12:
no
>0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)

nfp0, 2022-05-12:
My Bluetooth 4.0 TP-Link UB4A has that same signature. You sure that is not a Bluetooth 4.0 dongle?