nicman23/dkms-hid-nintendo #64: cannot build on armbian ubuntu

EricYin, 2023-04-09:
I have orangepi5 board with armbian linux installed (Armbian 23.02 Jammy Gnome, kernel 5.10.110). but when i run "sudo dkms build nintendo -v 3.2" on this orangepi 5, i got error below. is anyone know how i can fix it? I installed linux-headers-legacy-rockchip-rk3588_23.02.2_arm64.deb to build this driver before i got this error.

DKMS make.log for nintendo-3.2 for kernel 5.10.110-rockchip-rk3588 (aarch64)
Sat Apr  8 01:47:55 PM CST 2023
make: Entering directory '/usr/src/linux-headers-5.10.110-rockchip-rk3588'
  CC [M]  /var/lib/dkms/nintendo/3.2/build/src/hid-nintendo.o
  MODPOST /var/lib/dkms/nintendo/3.2/build/src/Module.symvers
make[2]: *** No rule to make target 'scripts/module.lds', needed by '/var/lib/dkms/nintendo/3.2/build/src/hid-nintendo.ko'.  Stop.
make[2]: *** Waiting for unfinished jobs....
  CC [M]  /var/lib/dkms/nintendo/3.2/build/src/hid-nintendo.mod.o
make[1]: *** [scripts/Makefile.modpost:175: __modpost] Error 2
make: *** [Makefile:1822: modules] Error 2
make: Leaving directory '/usr/src/linux-headers-5.10.110-rockchip-rk3588'

nicman23, 2023-04-10:
i would suggest just going with a backport kernel as upstream includes this driver. also i do not have an arm system to test anything on.

EricYin, 2023-04-11:
> i would suggest just going with a backport kernel as upstream includes this driver. also i do not have an arm system to test anything on.

Thanks, I figured ou the module.lds missing is a bug of armbian build code. I switch to use its master build code and get module.lds. then I can build and install nintendo hid driver. and also installed joycond with virtual gamepad mode on. my switch pro controller can be paired in my ubuntu arm OS, that is great. but I am still trying to figure out why moonlight streaming client can not make up/down/left/right button work when playing the game. any idea?

EricYin, 2023-04-15:
some update here: I bought new bluetooth adapter (real csr8510 chip). it is better than old one. but it seem after several minunites( playing the game), connection is dropped. the log of bluetooth said: host is down, cannot read hidinfo, something like this. the switch pro controller works great in windows 10. so after several days effort, I have to say hard to play with arm linux using swith pro controller.

nfp0, 2023-04-15:
> connection is dropped

@EricYin It's a known issue that it's trying to be solved. It happens during rumble.
You might want to subscribe to this issue from the kernel: https://github.com/DanielOgorchock/linux/issues/33

EricYin, 2023-04-16:
> > connection is dropped
> 
> @EricYin It's a known issue that it's trying to be solved. It happens during rumble. You might want to subscribe to this issue from the kernel: [DanielOgorchock/linux#33](https://github.com/DanielOgorchock/linux/issues/33)

Thanks, rumble is impartant for game, hope this can be fixed.

nicman23, 2023-04-16:
one workaround is to play with the controller plugged in

nfp0, 2023-04-16:
Another possible workaround is to use the controller through Steam, as the Steam driver does not seem to cause disconnects.

EricYin, 2023-04-19:
> does armbian have steam? wut

no, I did not see steam on armbian. I tried moonlight client streaming steam game from remote machine. lose bluetooth connection after several minutes..